EdwardLiuyc/StaticMapping #1: Problem about libstatic_mapping.so

carleft, 2019-08-26:
Hi,

When I make this project, there are some error with:
libstatic_mapping.so：undefined reference to ‘cv::imwrite(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, cv::_InputArray const&, std::vector<int, std::allocator<int> > const&)’
libstatic_mapping.so：undefined reference to ‘cv::_InputArray::_InputArray(cv::Mat const&)’
libstatic_mapping.so：undefined reference to ‘cv::line(cv::Mat&, cv::Point_<int>, cv::Point_<int>, cv::Scalar_<double> const&, int, int, int)’
collect2: error: ld returned 1 exit status
CMakeFiles/static_mapping_node.dir/build.make:577: recipe for target 'static_mapping_node' failed
make[2]: *** [static_mapping_node] Error 1
CMakeFiles/Makefile2:68: recipe for target 'CMakeFiles/static_mapping_node.dir/all' failed
make[1]: *** [CMakeFiles/static_mapping_node.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2

How can I solve it?

EdwardLiuyc, 2019-08-27:
add 
```
target_link_libraries(${TARGET_LIB_NAME} ${OpenCV_LIBRARIES})
```
after the `add_library` line in CMakeLists.txt

carleft, 2019-08-27:
> @carleft
> add
> 
> ```
> target_link_libraries(${TARGET_LIB_NAME} ${OpenCV_LIBRARIES})
> ```
> 
> after the `add_library` line in CMakeLists.txt
> 
> will commit soon!

Thank a lot, it works!

closed 2019-08-27 by carleft