EdwardLiuyc/StaticMapping #2: Question about tf connection and saving map

carleft, 2019-08-28:
Hi, I have some trouble when using the program.

First I tried to map only using pointcloud message, then I modified the mapping.sh :

## usally, you can just leave this config file just like this, it will work fine
CONFIG_PATH=./config/static_mapping_default.xml
## the follow 2 items must be set!!!
## the topic name of your pointcloud msg (ros)
POINT_CLOUD_TOPIC=/lslidar_point_cloud
## the frame id of your pointcloud msg (ros)
POINT_CLOUD_FRAME_ID=laser_link

./build/static_mapping_node \
  -cfg ${CONFIG_PATH} \
  -pc ${POINT_CLOUD_TOPIC} \
  -pc_frame_id ${POINT_CLOUD_FRAME_ID} \
  #-imu ${IMU_TOPIC} \
  #-imu_frame_id ${IMU_FRAME_ID} \
  #-odom ${ODOM_TOPIC} \
  #-odom_frame_id ${ODOM_FRAME_ID} \
  #-gps ${GPS_TOPIC} \
  #-gps_frame_id ${GPS_FRAME_ID} 

and there was no bug when running:

[ map_builder.cc: 921 ] [DEBUG] Add a new submap : 106
[ map_builder.cc: 921 ] [DEBUG] Add a new submap : 107
[ map_builder.cc: 921 ] [DEBUG] Add a new submap : 108
[ map_builder.cc: 190 ] [INFO] Got 500 clouds already.
[ map_builder.cc: 921 ] [DEBUG] Add a new submap : 109
[ map_builder.cc: 921 ] [DEBUG] Add a new submap : 110
[ map_builder.cc: 921 ] [DEBUG] Add a new submap : 111
[ map_builder.cc: 921 ] [DEBUG] Add a new submap : 112
[ map_builder.cc: 921 ] [DEBUG] Add a new submap : 113
[ map_builder.cc: 921 ] [DEBUG] Add a new submap : 114
[ map_builder.cc: 921 ] [DEBUG] Add a new submap : 115
[ map_builder.cc: 921 ] [DEBUG] Add a new submap : 116
[ map_builder.cc: 921 ] [DEBUG] Add a new submap : 117
[ map_builder.cc: 921 ] [DEBUG] Add a new submap : 118
[ map_builder.cc: 921 ] [DEBUG] Add a new submap : 119
^C[ map_builder.cc: 957 ] [INFO] Finishing Remaining Computations...
[ map_builder.cc: 450 ] [INFO] point cloud thread exit.
[ map_builder.cc: 899 ] [INFO] no enough frames for new submap, quit
[ map_builder.cc: 953 ] [INFO] submap processing done.

but I couldn't find the .pcd file in the pcd folder.

Another question is about tf connection, I noticed a warning in the mapping.sh that:

 If you got one of these topics, you MUST provide the tf connection between the one to pointcloud frame.

How can I provide the tf connection? Should I start up a rosnode like this? :

<launch>
	<node pkg="tf" type="static_transform_publisher" name="imu_to_pc"  args="0 0 0 0 0 0 /imu /laser_link 10" />
</launch>

Thank you very much in advance!

EdwardLiuyc, 2019-08-28:
@carleft 
Do you follow the installation tips on GTSAM in README? It is very important!
```
### important
# set GTSAM_USE_SYSTEM_EIGEN to ON
# then generate makefile
```
If not, set this option and rebuild&reinstall your gtsam, then rebuild this project!

carleft, 2019-08-28:
> @carleft
> It is because that your back-end(isam2) has been blocked!
> Do you follow the installation tips on GTSAM in README? It is very important!
> 
> ```
> ### important
> # set GTSAM_USE_SYSTEM_EIGEN to ON
> # then generate makefile
> ```
> 
> If not, set this option and rebuild&reinstall your gtsam, then rebuild this project!

Yes, I rebuild and ruinstall gtsam, then rebuild this project, but it seem useless.

carleft, 2019-08-29:
> @carleft
> It is because that your back-end(isam2) has been blocked!
> Do you follow the installation tips on GTSAM in README? It is very important!
>
> ```
> ### important
> # set GTSAM_USE_SYSTEM_EIGEN to ON [...]

I have solved this problem by modifying some code in the static_mapping_default.xml :

`
    <map_package_options
      enable="true"

      <imu_options 
        use_imu="true"
        imu_frequency="100" />

      <isam_optimizer_options 
        use_odom="false"
        use_gps="false" />

      <loop_detector_setting 
        use_gps="false"
        use_descriptor="false"

`
I am using pointcloud and imu msg,  and not using gps and odom msg.

EdwardLiuyc, 2019-08-29:
@carleft 
If your bags do not provide ` /tf` or `/tf_static` topic, I assume that you can just use `static_transform_publisher` to publish the tf topic instead.
If still not working, maybe you can do something like (just guess)
- rosbag play your_bag.bag --clock 
- set `use_sim_clock` to true

carleft, 2019-08-29:
> @carleft
> If your bags do not provide ` /tf` or `/tf_static` topic, I assume that you can just use `static_transform_publisher` to publish the tf topic instead.
> If still not working, maybe you can do something like (just guess)
> 
> * rosbag play your_bag.bag --clock
> * set `use_sim_clock` to true

Thanks a lot! It's very useful~

EdwardLiuyc, 2019-08-30:
@carleft 
Let me know if you still not working! 
Closing the problem.

closed 2019-08-30 by EdwardLiuyc