EdwardLiuyc/StaticMapping #5: compile error

droneboxid, 2019-11-27:
hi - please let me know why compile error on isam_optimizer.cc

droneboxid, 2019-11-27:
/StaticMapping/back_end/isam_optimizer.cc:199:18: error: ‘transform_from_’ is not a member of ‘gtsam’
           gtsam::transform_from_(gtsam::compose(Pose3_(GPS_COORD_KEY),  // map origin in GPS coord

droneboxid, 2019-11-27:
StaticMapping/back_end/isam_optimizer.cc:199:18: error: ‘transform_from’ is not a member of ‘gtsam’
           gtsam::transform_from(

insaneyilin, 2019-12-24:
> StaticMapping/back_end/isam_optimizer.cc:199:18: error: ‘transform_from’ is not a member of ‘gtsam’
> gtsam::transform_from(

I have met the same problem, checking out `4.0.0-alpha2` tag of GTSAM solves it.

closed 2020-01-07 by EdwardLiuyc