EdwardLiuyc/StaticMapping #7: Issue about tf between lidar and base_link

ahandsomeperson, 2020-04-09:
Thanks for your great work!
But when I run this program, it keeps waiting for tf between base_link and velodyne frame
[image]
This is my mapping.sh:
CONFIG_PATH=./config/static_mapping_default.xml
URDF_FILE=./urdf/test.urdf
POINT_CLOUD_TOPIC=/velodyne_points
POINT_CLOUD_FRAME_ID=velodyne
./build/static_mapping_node \
  -cfg ${CONFIG_PATH} \
  -pc ${POINT_CLOUD_TOPIC} \
  -pc_frame_id ${POINT_CLOUD_FRAME_ID} \
  #-imu ${IMU_TOPIC} \
  #-imu_frame_id ${IMU_FRAME_ID}
   -urdf ${URDF_FILE} \
  # -odom ${ODOM_TOPIC} \
  # -odom_frame_id ${ODOM_FRAME_ID} \
  # -gps ${GPS_TOPIC} \
  # -gps_frame_id ${GPS_FRAME_ID} \

exit 0

Then how can I fix this situation?

EdwardLiuyc, 2020-04-09:
it works for me!
maybe you should remove all commented lines after `./build/static_mapping_node \`and never forget the ' \ ' at the end of each line.

```
CONFIG_PATH=./config/static_mapping_default.xml  
URDF_FILE=./urdf/test.urdf  
POINT_CLOUD_TOPIC=/velodyne_points  
POINT_CLOUD_FRAME_ID=velodyne
./build/static_mapping_node \
  -cfg ${CONFIG_PATH} \
  -pc ${POINT_CLOUD_TOPIC} \
  -pc_frame_id ${POINT_CLOUD_FRAME_ID} \
  -urdf ${URDF_FILE} \
exit 0
```

ahandsomeperson, 2020-04-09:
Thank you ! Now I set POINT_CLOUD_FRAME_ID=base_link, it seems running.
But when I kill the program, the teminal shows:
[image]
No pointcloud and no graph, so the global map is not saved. So how can I fix this?

EdwardLiuyc, 2020-04-09:
Just like it said, no pointcloud was inserted. 
You may check your rosbag or lidar node for whether the pointcloud messages are correctly sent and receieved.
And I remeber that the latest version needs imu, If you want do mapping with only lidar, you can try the some older commited version, for example, "first commit".

closed 2020-04-26 by EdwardLiuyc