EdwardLiuyc/StaticMapping #8: segmentation fault

vijaygovindappa, 2020-04-13:
Hi, 
First of all amazing work and thank you for sharing. 
I have followed  all the steps and changed the topic names as relative to my rosbag. When i run the node. It will run for some time and i will get the segmentation fault. PFA
1. With IMU, i am getting the segmentation fault. PFA 
[image].
Waiting for your reply.

EdwardLiuyc, 2020-04-14:
Hello @vijaygovindappa 
I noticed that your imu data is without gravity, maybe you should send the raw imu data instead.

But I do not think this is the cause of the seg-fault, maybe you can change the cmakefile to build the debug version binary and use GDB to get the specific line where the seg-fault happens.

Still if you can upload your configs and rosbag, I may help you to debug.

vijaygovindappa, 2020-04-15:
Hello @EdwardLiuyc , 
Thank you for fast response. I do not have IMU raw data in my bag. 
I have ran the node in gdb and attached the results. I am sorry , i do not think i have the authority to share the rosbag. I would appreciate if we can debug this together. 
Thank you. 
[image]

EdwardLiuyc, 2020-04-15:
One possible cause is that your gtsam is using its own Eigen. It conficts with the eigen in system. You should follow the pipeline of building in README especially for GTSAM (using eigen in system);

Still, the information is not enough, you can upload the backtrace of the seg-fault. (`bt` in gdb)

vijaygovindappa, 2020-04-15:
I have made sure i am using system EIgen. The Eigen version is 3.2.92. I have attached the screenshot as well. I have also attached the screenshot of the backtrace
[image]
[image]

EdwardLiuyc, 2020-04-16:
the backrace shows that the seg-fault happens at icp aligning using libpointmatcher, I have no idea adout it because it never happens to me. 

maybe refer to https://github.com/ethz-asl/libpointmatcher/issues for some information.

by the way, I use the commit "eb9457db0bcea39367e7eebb4fe4c6974dd7f367" of libpointmatcher.

vijaygovindappa, 2020-04-16:
Okay. Thank you. I will update once i fix this issue. BTW i have one question. Is it mandatory to have eigen 3.3  or any version of eigen 3 works?

EdwardLiuyc, 2020-04-16:
I myself use eigen 3.2.92 too. 
Got it with apt-get in Ubuntu 16.04.

EdwardLiuyc, 2020-09-14:
re-open this if you still have the problem.

closed 2020-09-14 by EdwardLiuyc