EdwardLiuyc/StaticMapping #33: No odom/GPS data accessed!

ChienLiu, 2020-09-26:
Hello, 

I met some problem
when I runing the mapping sh file
I got warning like
"No odom data accessed!"
"No GPS data accessed!"

Then, I found these messages in "static_mapping_node.cpp"
[image]

I sure that I set the correct GPS topic name and the GPS frame ID
[image]
[image]

How can i fix it？
Thanks for your answer!

EdwardLiuyc, 2020-09-27:
Hello there, I have some tips that might be helpful: 
1. I **do not recommend** you to use odom only if you get very accuray odometry from a INS&RTK module. The same for GPS msg.
2. Maybe, you can try `mappling_lidar_only.sh` at first to check if you can get a reasonable map.
I haven't use GPS or odom msg in mapping for a long time, I will check if the functions still work properly and give you more informations later.

EdwardLiuyc, 2020-09-28:
@ChienLiu 
Hello again,  It works for me with a rosbag that includes gps data. You may have run a wrong script, since the `mapping.sh` do not use gps by default. as you can see that they are commented.
```
  # -urdf ${URDF_FILE} \
  # -odom ${ODOM_TOPIC} \
  # -odom_frame_id ${ODOM_FRAME_ID} \
  # -gps ${GPS_TOPIC} \
  # -gps_frame_id ${GPS_FRAME_ID} \
```
Or maybe you have fotgotten to add a `\` at the end of `-imu_frame_id ${IMU_FRAME_ID}`.

ChienLiu, 2020-09-29:
Thanks for your answer,
I'll check again.

closed 2020-09-29 by ChienLiu