EdwardLiuyc/StaticMapping #46: Get stuck in finding icp init guess for loop closure

ahandsomeperson, 2021-01-27:
Hi Liu! 
    Thanks for your great job!
    I am trying to find an icp init guess in loop closure based on lego loam. However i get stuck and it performs bad.
    After I read your _bool LoopDetector::CloseLoop_ function in loop_detector.cc, I find we are in a same way to add init guess for icp. As you comment in your code, it is a bug and it is quite wrong to simply calculating like this. Have you fixed this bug? Do you have any advice on this problem?

EdwardLiuyc, 2021-01-28:
Hi handsome!
The commented code is for gps and the calculation like that is wrong indeed, which simply uses the delta translation in gps but ignores the influence from rotation. so, my advice: 
1. Use a gps device which provides you with 6DOF global pose. That makes more sense for add init guess for icp.
2. The backend in this project will do the graph optimization using gps as a factor(only translation part) all the time, if so, the global pose you got will be rather accurate and you do not need any modification on the init guess. This is what I did in this project, otherwise, a better global match algorithm may be needed.

ahandsomeperson, 2021-01-28:
Thank you very much!

closed 2021-01-28 by ahandsomeperson