EdwardLiuyc/StaticMapping #47: .mapping.sh generates symbol lookup error  undefined symbol

Rotoslider, 2021-02-18:
I have the following error when running `./mapping.sh` from the StaticMapping folder:
```
lbackpack@lbackpack:/media/lbackpack/AGXSSD500/lidar/StaticMapping$ ./mapping.sh
./build/ros_node/static_mapping_node: symbol lookup error: /media/lbackpack/AGXSSD500/lidar/StaticMapping/build/back_end/libstatic_mapping_backend.so: undefined symbol: _ZNK5gtsam5Pose314transform_fromERKNS_6Point3ENS_16OptionalJacobianILi3ELi6EEENS4_ILi3ELi3EEE
```
System is a Jetson AGX arm64 with Ubuntu 18.04. pcl1-1-8 All dependencies are loaded. 
Did not use Docker.
make and make check both worked no errors.
I edited the mapping.sh to read my topics for the lidar and imu

When I run the` ./mapping.sh` is when the error happens.
Any suggestions?
thanks in advance

EdwardLiuyc, 2021-02-18:
It seems that `transform_from` has been removed in the later versions of gtsam. so I am not sure if you installed multiple version of gtsam. If not, may modify the `CMakelists.txt` to ensure the links are good.
Btw, `mapping.sh` has not been maintained for a long time, `mapping_lidar_imu.sh` is more recommended if you use lidar&imu. And better read https://github.com/EdwardLiuyc/StaticMapping/wiki at first.

Feel free to contact me if you have any further questions.

EdwardLiuyc, 2021-02-18:
Futhermore, I am really curious whether this can be run on an arm platform like Jetson, so if you achievement more progress, let me know please!

Rotoslider, 2021-02-18:
Looks like `transform_from` is now `transformFrom`. 
I have gtsam 4.0.2 installed from before. then when running your installer scripts I ran` ./StaticMapping/setup/install_gtsam.sh` which I assumed would update it to 4.0.3. 
I am very new to programming and Linux so I may have done something wrong. I will see if I can figure out what if anything I need to change in the CMakelists.txt file. 

thanks again

EdwardLiuyc, 2021-02-19:
So I assume that you have a conflict in multiple versions of gtsam. If the older version is not needed, may remove headers from `/usr/local/include/gtsam...` and libs called `/use/local/lib/libgtsam***.so` and reinstall the 4.0.3 version. Otherwise, you may change the install dir for new gtsam when `make install` and use CMake to find out the gtsam of specific version.

Rotoslider, 2021-02-20:
Thanks for the suggestions. I had already tired that once you mentioned I had two gtsam versions. I cleaned it out and then built and installed 4.0.3. I verified the header is calling for 4.0.3. And SC-LegoLoam uses it and it works fine with what I had before and the new 4.0.3.
Any other suggestions?

And thank you for taking the time to try and help. No so many do that anymore

Rotoslider, 2021-02-22:
Have you had time to try this with only 4.0.3 installed to see if you can repeat the error? Could there be another dependency I am missing? Or just something calling for something not in the current version of GTSAM?

EdwardLiuyc, 2021-02-23:
Sorry for the late reply.
I may have been misleaded before, actually I am using the version 4.0.0-alpha2 for gtsam (you can see in the setup). and it's quite different with 4.0.2 or 4.0.3. so as for your situation, you must have 4.0.0-alpha2 & 4.0.3 at the same time.

My suggestion, there may be 2 ways you can try:
1. Install 4.0.0-alpha2 locally in this project space and modify the cmakelists to `find_package` locally instead of finding package in system directories.
2. OR Just clone gtsam in 4.0.0-alpha2 version into this project and use it as a thirdparty which build along with the entire project. No need for install.

Rotoslider, 2021-02-23:
I vote for #3. I'm not a programmer so I will probably not be successful but I can try. 
I only have so much room on this Jetson and its getting full. Another version of GTSAM will just clog it up even more. 
thanks again, Looking forward to seeing what happens

EdwardLiuyc, 2021-02-24:
OK, I am not on this project at this moment so you may need to wait for a few days. 
I will go back to you as long as it's done.

Rotoslider, 2021-02-25:
> 
> 
> OK, I am not on this project at this moment so you may need to wait for a few days.
> I will go back to you as long as it's done.

I looked and my skill set is is not good enough to fix it. If you find the time to fix I will give it a try.
thanks again, Much appreciated.

EdwardLiuyc, 2021-03-16:
Hi @Rotoslider 
Now the repo is using gtsam in 4.0.3 instead, you could have another try if you are still interested in it. Before that, you may reinstall gtsam using the script: `setup/install_gtsam.sh`

Rotoslider, 2021-03-16:
The new version built correctly. Thank you.

closed 2021-03-16 by Rotoslider