EdwardLiuyc/StaticMapping #50: Warning skip cloud imu / lidar issue

Rotoslider, 2021-03-16:
I am having an issue with my data. 
I am running a Xsens MTi IMU and a Velodyne VLP-16
No GPS or ODOM
I have setup the `mapping_lidar_imu.sh` start script to read my topics. I have verified the bag file is outputting the topics and the proper data by` rostopic echo imu_data `and `rostopic echo velodyne_points`
The program starts up like this:
```
lbackpack@lbackpack:/media/lbackpack/AGXSSD500/StaticMapping$ ./mapping_lidar_imu.sh
[ static_mapping_node.cpp: 112 ] [INFO] Get point cloud from ROS topic: /velodyne_points
[ static_mapping_node.cpp: 128 ] [INFO] Get imu data from ROS topic: /imu/data
[ static_mapping_node.cpp: 146 ] [WARNING] No odom data accessed!
[ static_mapping_node.cpp: 161 ] [WARNING] No GPS data accessed!
[ static_mapping_node.cpp: 183 ] [WARNING] -pubs does not exist.
[ map_builder.cc: 152 ] [INFO] Got tf : tracking -> lidar 
  translation : (0, 0, 0)
  rotation    : (0, -0, 0)
[ map_builder.cc: 130 ] [INFO] Got tf : tracking -> imu 
  translation : (0, 0, 0)
  rotation    : (0, -0, 0)
[ map_builder_options.cc: 103 ] [INFO] Load configurations from file: ./config/lidar_imu_default.xml

*********************** configs from file ***********************

                      whole_options.export_file_path -> pcd/
                      whole_options.map_package_path -> pkgs/test/
                       whole_options.odom_calib_mode -> [default] 1
                       whole_options.separate_output -> [default] false
            whole_options.output_direct_combined_map -> true
                           whole_options.output_mrvm -> false
                     whole_options.output_kitti_pose -> [default] false
                         whole_options.separate_step -> [default] 200

                 output_mrvm_settings.output_average -> false
                     output_mrvm_settings.output_rgb -> false
              output_mrvm_settings.use_max_intensity -> [default] true
                 output_mrvm_settings.prob_threshold -> 0.57
                 output_mrvm_settings.low_resolution -> 1
                output_mrvm_settings.high_resolution -> 0.1
                       output_mrvm_settings.hit_prob -> 0.55
                      output_mrvm_settings.miss_prob -> 0.48
                       output_mrvm_settings.z_offset -> 0
          output_mrvm_settings.max_point_num_in_cell -> 10

                              Creating filter: Range
                                          min_range_ -> 4.5
                                          max_range_ -> 75
                      Creating filter: RandomSampler
                                      sampling_rate_ -> 0.4

                                       options->type -> 6
                         options->accepted_min_score -> [default] 0.7
              front_end_options.accumulate_cloud_num -> 151
             motion_filter_options.translation_range -> 1
                   motion_filter_options.angle_range -> 1.5
                    motion_filter_options.time_range -> [default] 0
                  motion_compensation_options.enable -> true
             motion_compensation_options.use_average -> true
                                 imu_options.enabled -> true
                                    imu_options.type -> 0
                               imu_options.frequency -> 200
                        imu_options.gravity_constant -> 9.8
                                       options->type -> 6
                         options->accepted_min_score -> 0.75
                          submap_options.frame_count -> 2
                    submap_options.enable_inner_mrvm -> [default] false
                  submap_options.enable_voxel_filter -> false
              submap_options.enable_random_sampleing -> false
                 submap_options.random_sampling_rate -> 0.5
                           submap_options.voxel_size -> [default] 0.1
                   submap_options.enable_disk_saving -> true
                         submap_options.enable_check -> false
                    submap_options.disk_saving_delay -> 30
                   submap_options.saving_name_prefix -> s_
                     isam_optimizer_options.use_odom -> false
                      isam_optimizer_options.use_gps -> false
             isam_optimizer_options.output_graph_pic -> [default] false
       isam_optimizer_options.enable_extrinsic_calib -> [default] true
          isam_optimizer_options.gps_factor_init_num -> [default] 25
       isam_optimizer_options.gps_factor_sample_step -> [default] 1
    isam_optimizer_options.gps_factor_init_angle_rad -> [default] 1.6
                       loop_detector_setting.use_gps -> false
                loop_detector_setting.use_descriptor -> true
          loop_detector_setting.output_matched_cloud -> false
      loop_detector_setting.trying_detect_loop_count -> 1
         loop_detector_setting.loop_ignore_threshold -> 15
       loop_detector_setting.nearest_history_pos_num -> 5
       loop_detector_setting.max_close_loop_distance -> 3
     loop_detector_setting.max_close_loop_z_distance -> 2
              loop_detector_setting.m2dp_match_score -> 0.96
       loop_detector_setting.accept_scan_match_score -> 0.8
                          map_package_options.enable -> false
                   map_package_options.border_offset -> 100
                     map_package_options.piece_width -> 500
               map_package_options.cloud_file_prefix -> part_
               map_package_options.descript_filename -> map_package.xml

*****************************************************************

[ map_builder.cc: 63 ] [INFO] Init scan matchers.
       dist_outlier_ratio -> 0.7
      knn_normal_estimate -> 7
            max_iteration -> 100
[ map_builder.cc: 76 ] [INFO] Init isam optimizer.
[ map_builder.cc: 218 ] [INFO] Add a new trajectory : 0
[ map_builder.cc: 102 ] [INFO] Init threads.
[ map_builder.cc: 111 ] [INFO] Init finished.
[ map_builder.cc: 783 ] [INFO] disable odom.
[ map_builder.cc: 792 ] [INFO] disable gps.
[ memory_manager.cc: 53 ] [INFO] Start managing memory.
[ map_builder.cc: 162 ] [WARNING] skip cloud.
[ map_builder.cc: 162 ] [WARNING] skip cloud.
[ map_builder.cc: 162 ] [WARNING] skip cloud.
[ map_builder.cc: 162 ] [WARNING] skip cloud.
[ map_builder.cc: 162 ] [WARNING] skip cloud.
```
I have verified that the software works with a different bag than mine. It will save out the path and point cloud in the pcd folder when done.
It does not like something about my imu or points. It could be translation or rotation in the urdf file maybe or time. The IMU and lidar are setup to ROS 105 and the axis's are in line with each other.  The IMU sits directly under the lidar. I did not add the translation numbers yet. It should be close enough not to mess it up to much. 
If I run the lidar data only using the `mapping_lidar_only.sh` it finishes but the point cloud look like a big doughnut with the hole in the middle.
The same bag file in LegoLoam produces a wonderful point cloud. It does not use the IMU data though. 
Any suggestions on what I can check?

EdwardLiuyc, 2021-03-17:
1. the `skip cloud` is a normal behaviour in the initialization phase in you use lidar&imu. You can just ignore it if it does not last forever.
2. It's recommended to try `mapping_lidar_only.sh` at first. Find `lidar_only.xml` in config folder and change `accumulate_cloud_num="151"` to `accumulate_cloud_num="1"`, then you may get a more reasonable result.

Btw, could you upload your rosbag to Google Drive or some other cloud drive so that I can debug with your data simultaneously?

Rotoslider, 2021-03-18:
Thank you for the suggestions. I missed the change `accumulate_cloud_num="151" to accumulate_cloud_num="1"` that was described in the wiki. This allows the map to build better using the lidar_only script. Its still not usable but you can tell what things are now. 
It still fails completely using the lidar with IMU.
I sent you a link to the bag file to the address you provided yesterday. I also described the issues more thoroughly in the email.
Thank you again for your help.

Rotoslider, 2021-03-21:
Part of my issue was a time sync between the IMU and the Lidar. I had switched Lidar units and did not plug in the GPS antenna. The ROS Velodyne driver had "Use GPS time" set to true. You will need to set this to false if you want to run without a GPS antenna.
I still have not been able to produce a good map but by fixing the time sync issue the bag files will now compute in StaticMapping.  
More testing to go but Skip Cloud issue has been solved thanks to Edward.

closed 2021-03-21 by Rotoslider