EdwardLiuyc/StaticMapping #54: Failing at values.empty() check in icp_fast

ibrahimhroob, 2021-04-26:
Hi @EdwardLiuyc 

I think I have set my topic + bag_file_dir correct in "mapping_lidar_only.sh", but when I run it, it got failed at !values.empty() [**F0426 18:09:24.934926 23185 icp_fast.cc:81] Check failed: !values.empty()** ]

Please find the attached output below. 

Hope to hear from you soon. 

Many thanks, 

===============================================================================

ibrahim@ihroob:~/StaticMapping/src/StaticMapping$ ./mapping_lidar_only.sh
[ static_mapping_node.cpp: 118 ] [INFO] Get point cloud from ROS topic: /points_raw
[ static_mapping_node.cpp: 132 ] [WARNING] No imu topic, expect no imu messages.
[ static_mapping_node.cpp: 152 ] [WARNING] No odom data accessed!
[ static_mapping_node.cpp: 167 ] [WARNING] No GPS data accessed!
[ map_builder.cc: 152 ] [INFO] Got tf : tracking -> lidar 
  translation : (0, 0, 0)
  rotation    : (0, -0, 0)
[ map_builder_options.cc: 103 ] [INFO] Load configurations from file: ./config/lidar_only.xml

*********************** configs from file ***********************

                      whole_options.export_file_path -> pcd/
                      whole_options.map_package_path -> pkgs/test/
                       whole_options.odom_calib_mode -> [default] 1
                       whole_options.separate_output -> [default] false
            whole_options.output_direct_combined_map -> true
                           whole_options.output_mrvm -> false
                     whole_options.output_kitti_pose -> [default] false
                         whole_options.separate_step -> [default] 200

                 output_mrvm_settings.output_average -> false
                     output_mrvm_settings.output_rgb -> false
              output_mrvm_settings.use_max_intensity -> [default] true
                 output_mrvm_settings.prob_threshold -> 0.57
                 output_mrvm_settings.low_resolution -> 1
                output_mrvm_settings.high_resolution -> 0.1
                       output_mrvm_settings.hit_prob -> 0.55
                      output_mrvm_settings.miss_prob -> 0.48
                       output_mrvm_settings.z_offset -> 0
          output_mrvm_settings.max_point_num_in_cell -> 10

                              Creating filter: Range
                                          min_range_ -> 4.5
                                          max_range_ -> 75
                      Creating filter: RandomSampler
                                      sampling_rate_ -> 0.5

                                       options->type -> 6
                         options->accepted_min_score -> [default] 0.7
              front_end_options.accumulate_cloud_num -> 151
             motion_filter_options.translation_range -> 1
                   motion_filter_options.angle_range -> 1.5
                    motion_filter_options.time_range -> 0.25
                  motion_compensation_options.enable -> true
             motion_compensation_options.use_average -> true
                                 imu_options.enabled -> false
                                    imu_options.type -> 0
                               imu_options.frequency -> 200
                        imu_options.gravity_constant -> 9.8
                                       options->type -> 6
                         options->accepted_min_score -> 0.7
                          submap_options.frame_count -> 2
                    submap_options.enable_inner_mrvm -> [default] false
                  submap_options.enable_voxel_filter -> false
              submap_options.enable_random_sampleing -> false
                 submap_options.random_sampling_rate -> 0.5
                           submap_options.voxel_size -> [default] 0.1
                   submap_options.enable_disk_saving -> true
                         submap_options.enable_check -> false
                    submap_options.disk_saving_delay -> 30
                   submap_options.saving_name_prefix -> s_
                     isam_optimizer_options.use_odom -> false
                      isam_optimizer_options.use_gps -> false
             isam_optimizer_options.output_graph_pic -> [default] false
       isam_optimizer_options.enable_extrinsic_calib -> [default] true
          isam_optimizer_options.gps_factor_init_num -> [default] 25
       isam_optimizer_options.gps_factor_sample_step -> [default] 1
    isam_optimizer_options.gps_factor_init_angle_rad -> [default] 1.6
                       loop_detector_setting.use_gps -> false
                loop_detector_setting.use_descriptor -> true
          loop_detector_setting.output_matched_cloud -> false
      loop_detector_setting.trying_detect_loop_count -> 1
         loop_detector_setting.loop_ignore_threshold -> 20
       loop_detector_setting.nearest_history_pos_num -> 5
       loop_detector_setting.max_close_loop_distance -> 5
     loop_detector_setting.max_close_loop_z_distance -> 2
              loop_detector_setting.m2dp_match_score -> 0.975
       loop_detector_setting.accept_scan_match_score -> 0.8
                          map_package_options.enable -> false
                   map_package_options.border_offset -> 100
                     map_package_options.piece_width -> 500
               map_package_options.cloud_file_prefix -> part_
               map_package_options.descript_filename -> map_package.xml

*****************************************************************

[ map_builder.cc: 63 ] [INFO] Init scan matchers.
       dist_outlier_ratio -> 0.7
      knn_normal_estimate -> 7
            max_iteration -> 100
[ map_builder.cc: 76 ] [INFO] Init isam optimizer.
[ map_builder.cc: 218 ] [INFO] Add a new trajectory : 0
[ map_builder.cc: 102 ] [INFO] Init threads.
[ map_builder.cc: 111 ] [INFO] Init finished.
[ map_builder.cc: 783 ] [INFO] disable odom.
[ map_builder.cc: 792 ] [INFO] disable gps.
[ memory_manager.cc: 53 ] [INFO] Start managing memory.
F0426 18:09:24.934926 23185 icp_fast.cc:81] Check failed: !values.empty() 
*** Check failure stack trace: ***
    @     0x7f334bd2e0cd  google::LogMessage::Fail()
    @     0x7f334bd2ff33  google::LogMessage::SendToLog()
    @     0x7f334bd2dc28  google::LogMessage::Flush()
    @     0x7f334bd30999  google::LogMessageFatal::~LogMessageFatal()
    @     0x7f334d5d6546  static_map::registrator::IcpFast::Align()
    @     0x7f334de0accb  static_map::MapBuilder::ScanMatchProcessing()
    @     0x7f334ba566df  (unknown)
    @     0x7f334b1786db  start_thread
    @     0x7f334b4b171f  clone
./mapping_lidar_only.sh: line 32: 23177 Aborted                 (core dumped) ./build/ros_node/static_mapping_node -cfg ${CONFIG_PATH} -pc ${POINT_CLOUD_TOPIC} -pc_frame_id ${POINT_CLOUD_FRAME_ID} -urdf ${URDF_FILE} -bag ${BAG_FILE} -pubs [map][path][edge]

Rotoslider, 2021-04-26:
Have you ran `rostopic list -v` to see a list of the topics while playing the bag file to make sure the topics and frame id in your bag file match the ones in mapping_lidar_only.sh?
Have you verified the bag file topics are outputting data with `rostopic echo velodyne_points` for example. 
You may want to use mapping_lidar_imu.sh.

ibrahimhroob, 2021-04-26:
Many thanks for your quick reply (Y) 

It seems the issue that I had was with the topic name, it did not like having '/' before topic name. I have just set the topic name and that was much better. Please find the photo below for my "mapping_lidar_only.sh" file just to make sure I am not missing something obvious! 
[image]

Here what I got when `rostopic list -v`: 
[image]

But there was no output from the topic when I was trying to echo the topic while the algorithm is running. However, I am sure the bag file is not empty, here what I got when I display the bag information: 
[image]

===================================================================
This is the output after I fixed the topic name: 

`ibrahim@ihroob:~/StaticMapping/src/StaticMapping$ ./mapping_lidar_only.sh 
[ static_mapping_node.cpp: 118 ] [INFO] Get point cloud from ROS topic: points_raw
[ static_mapping_node.cpp: 132 ] [WARNING] No imu topic, expect no imu messages.
[ static_mapping_node.cpp: 152 ] [WARNING] No odom data accessed!
[ static_mapping_node.cpp: 167 ] [WARNING] No GPS data accessed!
[ map_builder.cc: 152 ] [INFO] Got tf : tracking -> lidar 
  translation : (0, 0, 0)
  rotation    : (0, -0, 0)
[ map_builder_options.cc: 103 ] [INFO] Load configurations from file: ./config/lidar_only.xml

*********************** configs from file ***********************

                      whole_options.export_file_path -> pcd/
                      whole_options.map_package_path -> pkgs/test/
                       whole_options.odom_calib_mode -> [default] 1
                       whole_options.separate_output -> [default] false
            whole_options.output_direct_combined_map -> true
                           whole_options.output_mrvm -> true
                     whole_options.output_kitti_pose -> [default] false
                         whole_options.separate_step -> [default] 200

                 output_mrvm_settings.output_average -> false
                     output_mrvm_settings.output_rgb -> false
              output_mrvm_settings.use_max_intensity -> [default] true
                 output_mrvm_settings.prob_threshold -> 0.57
                 output_mrvm_settings.low_resolution -> 1
                output_mrvm_settings.high_resolution -> 0.1
                       output_mrvm_settings.hit_prob -> 0.55
                      output_mrvm_settings.miss_prob -> 0.48
                       output_mrvm_settings.z_offset -> 0
          output_mrvm_settings.max_point_num_in_cell -> 10

                              Creating filter: Range
                                          min_range_ -> 4.5
                                          max_range_ -> 75
                      Creating filter: RandomSampler
                                      sampling_rate_ -> 0.5

                                       options->type -> 6
                         options->accepted_min_score -> [default] 0.7
              front_end_options.accumulate_cloud_num -> 151
             motion_filter_options.translation_range -> 1
                   motion_filter_options.angle_range -> 1.5
                    motion_filter_options.time_range -> 0.25
                  motion_compensation_options.enable -> true
             motion_compensation_options.use_average -> true
                                 imu_options.enabled -> false
                                    imu_options.type -> 0
                               imu_options.frequency -> 200
                        imu_options.gravity_constant -> 9.8
                                       options->type -> 6
                         options->accepted_min_score -> 0.7
                          submap_options.frame_count -> 2
                    submap_options.enable_inner_mrvm -> [default] false
                  submap_options.enable_voxel_filter -> false
              submap_options.enable_random_sampleing -> false
[... 36 lines cut ...]
[ map_builder.cc: 218 ] [INFO] Add a new trajectory : 0
[ map_builder.cc: 102 ] [INFO] Init threads.
[ map_builder.cc: 111 ] [INFO] Init finished.
[ map_builder.cc: 783 ] [INFO] disable odom.
[ map_builder.cc: 792 ] [INFO] disable gps.
[ memory_manager.cc: 53 ] [INFO] Start managing memory.
processed 50.815000 of 163.600000 seconds of the bag... 
processed 100.620000 of 163.600000 seconds of the bag... 
processed 150.535000 of 163.600000 seconds of the bag... 
[ map_builder.cc: 716 ] [INFO] Finishing Remaining Computations...
[ map_builder.cc: 396 ] [INFO] point cloud thread exit.
[ map_builder.cc: 665 ] [INFO] no enough frames for new submap, quit
[ map_builder.cc: 712 ] [INFO] submap processing done.
[ map_builder.cc: 566 ] [INFO] All submaps have been connected.
[ map_builder.cc: 616 ] [INFO] generating path files ...
[ map_package.cc: 93 ] [WARNING] No good bounding box, save no map package.
[ map_builder.cc: 898 ] [INFO] creating the static map ...
[ multi_resolution_voxel_map.cc: 238 ] [INFO] Finished filtering output cloud, generating pcd file.
[ multi_resolution_voxel_map.cc: 238 ] [WARNING] Cloud is empty. Do not output to file.
[ memory_manager.cc: 62 ] [INFO] End managing memory.

               block name |    times |  avg.(ms) |   sum.(ms) |  min.(ms) |  max.(ms)
-------------------------------------------------------------------------------------
                     main |        1 |     19001 |      19001 |     19001 |     19001
`

EdwardLiuyc, 2021-04-27:
Hi @ibrahimhroob 
I think the problem may caused by this parameter: `front_end_options.accumulate_cloud_num`, you may find this parameter in the config `.xml` file and change it from 151 to 1.
If you still run into troubles, do not hesitate to reach out to me again.

ibrahimhroob, 2021-04-27:
Hi @EdwardLiuyc 

I have set 'accumulate_cloud_num' param to 1, please find the output below. 
Not sure where to look at after it finishes, how I can see the output map? I have checked the **pcd** folder, it only contains path.csv and it was empty. 

Do I need to play rosbag explicitly while the algorithm is running? Is not that being handled internally? 

Thanks,  

ibrahim@ihroob:~/StaticMapping/src/StaticMapping$ ./mapping_lidar_only.sh 
[ static_mapping_node.cpp: 112 ] [INFO] Get point cloud from ROS topic: points_raw
[ static_mapping_node.cpp: 126 ] [WARNING] No imu topic, expect no imu messages.
[ static_mapping_node.cpp: 146 ] [WARNING] No odom data accessed!
[ static_mapping_node.cpp: 161 ] [WARNING] No GPS data accessed!
[ map_builder.cc: 152 ] [INFO] Got tf : tracking -> lidar 
  translation : (0.04, -0.02, 0.3)
  rotation    : (0, -0, 0)
[ map_builder_options.cc: 103 ] [INFO] Load configurations from file: ./config/lidar_only.xml

*********************** configs from file ***********************

                      whole_options.export_file_path -> pcd/
                      whole_options.map_package_path -> pkgs/test/
                       whole_options.odom_calib_mode -> [default] 1
                       whole_options.separate_output -> [default] false
            whole_options.output_direct_combined_map -> true
                           whole_options.output_mrvm -> false
                     whole_options.output_kitti_pose -> [default] false
                         whole_options.separate_step -> [default] 200

                 output_mrvm_settings.output_average -> false
                     output_mrvm_settings.output_rgb -> false
              output_mrvm_settings.use_max_intensity -> [default] true
                 output_mrvm_settings.prob_threshold -> 0.57
                 output_mrvm_settings.low_resolution -> 1
                output_mrvm_settings.high_resolution -> 0.1
                       output_mrvm_settings.hit_prob -> 0.55
                      output_mrvm_settings.miss_prob -> 0.48
                       output_mrvm_settings.z_offset -> 0
          output_mrvm_settings.max_point_num_in_cell -> 10

                              Creating filter: Range
                                          min_range_ -> 4.5
                                          max_range_ -> 75
                      Creating filter: RandomSampler
                                      sampling_rate_ -> 0.5

                                       options->type -> 6
                         options->accepted_min_score -> [default] 0.7
              front_end_options.accumulate_cloud_num -> 1
             motion_filter_options.translation_range -> 1
                   motion_filter_options.angle_range -> 1.5
                    motion_filter_options.time_range -> 0.25
                  motion_compensation_options.enable -> true
             motion_compensation_options.use_average -> true
                                 imu_options.enabled -> false
                                    imu_options.type -> 0
                               imu_options.frequency -> 200
                        imu_options.gravity_constant -> 9.8
                                       options->type -> 6
                         options->accepted_min_score -> 0.7
                          submap_options.frame_count -> 2
                    submap_options.enable_inner_mrvm -> [default] false
                  submap_options.enable_voxel_filter -> false
              submap_options.enable_random_sampleing -> false
                 submap_options.random_sampling_rate -> 0.5
                           submap_options.voxel_size -> [default] 0.1
                   submap_options.enable_disk_saving -> true
                         submap_options.enable_check -> false
                    submap_options.disk_saving_delay -> 30
[... 26 lines cut ...]
[ map_builder.cc: 63 ] [INFO] Init scan matchers.
       dist_outlier_ratio -> 0.7
      knn_normal_estimate -> 7
            max_iteration -> 100
[ map_builder.cc: 76 ] [INFO] Init isam optimizer.
[ map_builder.cc: 218 ] [INFO] Add a new trajectory : 0
[ map_builder.cc: 102 ] [INFO] Init threads.
[ map_builder.cc: 111 ] [INFO] Init finished.
[ map_builder.cc: 783 ] [INFO] disable odom.
[ map_builder.cc: 792 ] [INFO] disable gps.
[ memory_manager.cc: 53 ] [INFO] Start managing memory.
processed 50.815000 of 163.600000 seconds of the bag... 
processed 100.620000 of 163.600000 seconds of the bag... 
processed 150.535000 of 163.600000 seconds of the bag... 
[ map_builder.cc: 716 ] [INFO] Finishing Remaining Computations...
[ map_builder.cc: 396 ] [INFO] point cloud thread exit.
[ map_builder.cc: 665 ] [INFO] no enough frames for new submap, quit
[ map_builder.cc: 712 ] [INFO] submap processing done.
[ map_builder.cc: 566 ] [INFO] All submaps have been connected.
[ map_builder.cc: 616 ] [INFO] generating path files ...
[ memory_manager.cc: 62 ] [INFO] End managing memory.

               block name |    times |  avg.(ms) |   sum.(ms) |  min.(ms) |  max.(ms)
-------------------------------------------------------------------------------------
                     main |        1 |     19001 |      19001 |     19001 |     19001

EdwardLiuyc, 2021-04-27:
With these ouput, I can see that the point clouds are discarded, maybe due to the topic name or frame id you set.  
I think you can try with `/points_raw` again in the `.sh` file. And please make sure that your pointcloud messages have the frame id  which is set in the `.sh` file, in your case, is `base_link`.

ibrahimhroob, 2021-04-27:
Thank you, its all working now :+1:

closed 2021-04-27 by ibrahimhroob