EdwardLiuyc/StaticMapping #55: Output data format

ibrahimhroob, 2021-05-02:
Hi @EdwardLiuyc 

This might be a silly question but can you please confirm the output format of csv file? is it Time/Step, x, y, z, r, p, y? Is there away to bind or replace the step with time-stamp from the point cloud data? 

Thanks.

ibrahimhroob, 2021-05-03:
@EdwardLiuyc how I can save the trajectory/path in tum format? Thanks

EdwardLiuyc, 2021-05-04:
@ibrahimhroob 
1. it's index, x, y, z, r, p, y in path.csv
2. TUM format is not supported yet, only formats supported are KITTI format and self-defined format mentioned in 1.

If you want to change the index to timestamp, maybe you can modify `trajectory.cc` to make this happen. And if you want to get tum format path, you can refer to `OutputPath()` in `map_builder.cc`.

ibrahimhroob, 2021-05-04:
Many thanks for your reply, I have replaced 'path_content += (std::to_string(path_point_index) + ", ");' with 'path_content += (std::to_string(frame->GetTimeStamp().ToNanoSec()) + ", ");' (Y)

closed 2021-05-04 by ibrahimhroob