EdwardLiuyc/StaticMapping #59: Boost error while building

DaddyWesker, 2022-01-13:
Hello and thanks for your code!

Here is what i'm getting when i'm trying to build StaticMapping

```
/usr/bin/ld: ../libstatic_mapping_builder_data.so: undefined reference to `boost::this_thread::interruption_point()'
/usr/bin/ld: ../libstatic_mapping_builder_data.so: undefined reference to `boost::detail::get_current_thread_data()'
/usr/bin/ld: ../libstatic_mapping_builder_data.so: undefined reference to `boost::this_thread::disable_interruption::disable_interruption()'
/usr/bin/ld: ../libstatic_mapping_builder_data.so: undefined reference to `boost::this_thread::disable_interruption::~disable_interruption()'
collect2: error: ld returned 1 exit status
make[2]: *** [builder/data/test/CMakeFiles/test_cloud_types.dir/build.make:205: builder/data/test/test_cloud_types] Error 1
make[1]: *** [CMakeFiles/Makefile2:1064: builder/data/test/CMakeFiles/test_cloud_types.dir/all] Error 2
```
I was following README through all steps. Libboost is installed and mapping builder too. So i don't understand what is the problem there. ANy ideas?

DaddyWesker, 2022-01-13:
Without -DENABLE_TEST=ON make is working. But i guess this isn't suppose to be the problem?

EdwardLiuyc, 2022-01-13:
@DaddyWesker 
I will check this and come back to you once I get some conclusions.

EdwardLiuyc, 2022-02-01:
@DaddyWesker  Sorry to reply so late. This issue should already be fixed. You can update to the latest code.