EdwardLiuyc/StaticMapping #61: Question about running the code

DaddyWesker, 2022-01-20:
Hello again.

I've downloaded kitti sequence and converted it to rosbag so i can try to run it using your code. But i'm not exactly too familiar with robags, so i wanted to ask you about one thing. Here is the part of the `mapping_lidar_only.sh`. 

```
## usally, you can leave this config file just like this, it will work fine
CONFIG_PATH=./config/static_mapping_default.xml
## the follow 2 items must be set!!!
## the topic name of your pointcloud msg (ros)
POINT_CLOUD_TOPIC=/fused_point_cloud
## the frame id of your pointcloud msg (ros)
POINT_CLOUD_FRAME_ID=base_link
```

And here is the rosbag

```
path:        kitti_2011_09_26_drive_0011_synced.bag
version:     2.0
duration:    24.0s
start:       Sep 26 2011 13:10:04.78 (1317028204.78)
end:         Sep 26 2011 13:10:28.79 (1317028228.79)
size:        1.2 GB
messages:    3262
compression: none [932/932 chunks]
types:       geometry_msgs/TwistStamped [98d34b0043a2093cf9d9345ab6eef12e]
             sensor_msgs/CameraInfo     [c9a58c1b0b154e0e6da7578cb991d214]
             sensor_msgs/Image          [060021388200f6f0f447d0fcd9c64743]
             sensor_msgs/Imu            [6a62c6daae103f4ff57a132d6f95cec2]
             sensor_msgs/NavSatFix      [2d3a8cd499b9b4a0249fb98fd05cfa48]
             sensor_msgs/PointCloud2    [1158d486dd51d683ce2f1be655c3c181]
             tf2_msgs/TFMessage         [94810edda583a504dfda3829e70d7eec]
topics:      /kitti/camera_color_left/camera_info    233 msgs    : sensor_msgs/CameraInfo    
             /kitti/camera_color_left/image_raw      233 msgs    : sensor_msgs/Image         
             /kitti/camera_color_right/camera_info   233 msgs    : sensor_msgs/CameraInfo    
             /kitti/camera_color_right/image_raw     233 msgs    : sensor_msgs/Image         
             /kitti/camera_gray_left/camera_info     233 msgs    : sensor_msgs/CameraInfo    
             /kitti/camera_gray_left/image_raw       233 msgs    : sensor_msgs/Image         
             /kitti/camera_gray_right/camera_info    233 msgs    : sensor_msgs/CameraInfo    
             /kitti/camera_gray_right/image_raw      233 msgs    : sensor_msgs/Image         
             /kitti/oxts/gps/fix                     233 msgs    : sensor_msgs/NavSatFix     
             /kitti/oxts/gps/vel                     233 msgs    : geometry_msgs/TwistStamped
             /kitti/oxts/imu                         233 msgs    : sensor_msgs/Imu           
             /kitti/velo/pointcloud                  233 msgs    : sensor_msgs/PointCloud2   
             /tf                                     233 msgs    : tf2_msgs/TFMessage        
             /tf_static                              233 msgs    : tf2_msgs/TFMessage
```

If i've got it right, POINT_CLOUD_TOPIC is the /kitti/velo/pointcloud  in my case but what about POINT_CLOUD_FRAME_ID? I'm not quite sure what i need to put here.

If I got it right, in the step 2 where i need to play rosbag file, it should be something like this?
`rosbag play --clock kitti_2011_09_26_drive_0011_synced.bag`

Thanks in advance!

EdwardLiuyc, 2022-01-21:
Hi @DaddyWesker 
Actually the project supports to run directly with Kitti dataset. you can check `mapping_kitti.sh`, just change the path of dataset and run! 
Good luck with that!

DaddyWesker, 2022-01-26:
Here is what i've got using mapping_kitti.sh

```
[ static_mapping_node.cpp: 111 ] [INFO] Get point cloud from ROS topic: velodyne_points
[ static_mapping_node.cpp: 126 ] [WARNING] No imu topic, expect no imu messages.
[ static_mapping_node.cpp: 146 ] [WARNING] No odom data accessed!
[ static_mapping_node.cpp: 161 ] [WARNING] No GPS data accessed!
[ WARN] [1643193041.676134315]: link 'gps' material 'green' undefined.
[ WARN] [1643193041.676884604]: link 'gps' material 'green' undefined.
[ map_builder.cc: 152 ] [INFO] Got tf : tracking -> lidar 
  translation : (0.04, -0.02, 0.3)
  rotation    : (0, -0, 0)
[ map_builder_options.cc: 103 ] [INFO] Load configurations from file: ./config/lidar_only_kitti.xml

*********************** configs from file ***********************

                      whole_options.export_file_path -> pcd/
                      whole_options.map_package_path -> pkgs/test/
                       whole_options.odom_calib_mode -> [default] 1
                       whole_options.separate_output -> false
            whole_options.output_direct_combined_map -> false
                           whole_options.output_mrvm -> true
                     whole_options.output_kitti_pose -> true
                         whole_options.separate_step -> 200

                 output_mrvm_settings.output_average -> false
                     output_mrvm_settings.output_rgb -> true
              output_mrvm_settings.use_max_intensity -> true
                 output_mrvm_settings.prob_threshold -> 0.54
                 output_mrvm_settings.low_resolution -> 1
                output_mrvm_settings.high_resolution -> 0.1
                       output_mrvm_settings.hit_prob -> 0.55
                      output_mrvm_settings.miss_prob -> 0.49
                       output_mrvm_settings.z_offset -> 0
          output_mrvm_settings.max_point_num_in_cell -> 8

                              Creating filter: Range
                                          min_range_ -> 5
                                          max_range_ -> 3.40282e+38
                          Creating filter: AxisRange
                                                min_ -> -2
                                                max_ -> 3.40282e+38
                                         axis_index_ -> 2
                      Creating filter: RandomSampler
                                      sampling_rate_ -> 0.5

                                       options->type -> 6
                         options->accepted_min_score -> [default] 0.7
              front_end_options.accumulate_cloud_num -> 1
             motion_filter_options.translation_range -> 0.5
                   motion_filter_options.angle_range -> 1.5
                    motion_filter_options.time_range -> 0.25
                  motion_compensation_options.enable -> false
             motion_compensation_options.use_average -> true
                                 imu_options.enabled -> false
                                    imu_options.type -> 0
                               imu_options.frequency -> 200
                        imu_options.gravity_constant -> 9.8
                                       options->type -> 6
                         options->accepted_min_score -> 0.7
                          submap_options.frame_count -> 2
                    submap_options.enable_inner_mrvm -> [default] false
                  submap_options.enable_voxel_filter -> true
              submap_options.enable_random_sampleing -> false
                 submap_options.random_sampling_rate -> 0.5
                           submap_options.voxel_size -> 0.1
                   submap_options.enable_disk_saving -> true
                         submap_options.enable_check -> false
                    submap_options.disk_saving_delay -> 10
                   submap_options.saving_name_prefix -> s_
[... 33 lines cut ...]
[ map_builder.cc: 783 ] [INFO] disable odom.
[ map_builder.cc: 792 ] [INFO] disable gps.
[ static_mapping_node.cpp: 431 ] [INFO] No bag, read kitti data.
[ map_builder.cc: 716 ] [INFO] Finishing Remaining Computations...
[ memory_manager.cc: 53 ] [INFO] Start managing memory.
[ map_builder.cc: 396 ] [INFO] point cloud thread exit.
[ map_builder.cc: 665 ] [INFO] no enough frames for new submap, quit
[ map_builder.cc: 712 ] [INFO] submap processing done.
[ map_builder.cc: 566 ] [INFO] All submaps have been connected.
[ map_builder.cc: 616 ] [INFO] generating path files ...
[ map_builder.cc: 898 ] [INFO] creating the static map ...
[ multi_resolution_voxel_map.cc: 234 ] [INFO] Finished filtering output cloud, generating pcd file.
[ multi_resolution_voxel_map.cc: 234 ] [WARNING] Cloud is empty. Do not output to file.
[ memory_manager.cc: 62 ] [INFO] End managing memory.

               block name |    times |  avg.(ms) |   sum.(ms) |  min.(ms) |  max.(ms)
-------------------------------------------------------------------------------------
                     main |        1 |      1000 |       1000 |      1000 |      1000

```
This it KITTI_PATH i've used

`KITTI_PATH=/home/daddywesker/datasets/kitti/2011_09_29/2011_09_29_drive_0071_sync/velodyne_points`

Any ideas?

EdwardLiuyc, 2022-02-04:
Hi @DaddyWesker 
First you should set KITTI_PATH to `/home/daddywesker/datasets/kitti/2011_09_29/2011_09_29_drive_0071_sync/velodyne_points/data/`

And usually I use sequences in data_odometry_velodyne.zip for test, the file names inside are a little bit different to files in "2011_09_29_drive_0071_sync", so you need to change line 94 of `kitti_reader.cc` from 
`
snprintf(filename, sizeof(filename), "%s/%06d.bin", point_cloud_data_path_.c_str(), index);
`
to
`
snprintf(filename, sizeof(filename), "%s/%010d.bin", point_cloud_data_path_.c_str(), index);
`

Have fun!

EdwardLiuyc, 2022-02-04:
@DaddyWesker Or you can try with this branch: `flexiable_kitti_reader`
Just repleace the KITTI_PATH with your own path.

EdwardLiuyc, 2022-02-10:
@DaddyWesker This feature is merged to master already. Feel free to re-open this issue or create a new issue if you still get questions.

DaddyWesker, 2022-02-17:
Hello, @EdwardLiuyc . I was on vacation so haven't been able to test it out.

I've ran mapping_kitti.sh just fine after pulling new changes but where to find the map? I've pre-launched rviz but there is only current points on the screen and no map. 

Also, if i got it right, code is first collecting data and after sequence is ended it began to calculate map? So it is not online, but postprocessing?

EdwardLiuyc, 2022-02-17:
@DaddyWesker 
1. The map should be in ./pcd folder which is mentioned in README.
2. If you want to use RVIZ, you need a .rviz config file in config folder
3. No it is online. The postprocessing is for removing dynamic objects, if you want you can disable this postprocessing and directly output the map (there is actually a config parameter for this). I will add more info for config parameters in wiki.

closed 2022-03-04 by EdwardLiuyc