EdwardLiuyc/StaticMapping #65: Error cmake libpointmatcher

duongthao1218, 2022-03-03:
when i run `$ ./setup/install_libpointmatcher.sh`, I get error 
```
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[2]: *** [CMakeFiles/pointmatcher.dir/build.make:401: CMakeFiles/pointmatcher.dir/pointmatcher/ErrorMinimizers/PointToPlane.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/home/duongthao/catkin_ws/src/StaticMapping/libpointmatcher/pointmatcher/DataPointsFilters/CovarianceSampling.cpp: In instantiation of ‘void CovarianceSamplingDataPointsFilter<T>::inPlaceFilter(CovarianceSamplingDataPointsFilter<T>::DataPoints&) [with T = float; CovarianceSamplingDataPointsFilter<T>::DataPoints = PointMatcher<float>::DataPoints]’:
/home/duongthao/catkin_ws/src/StaticMapping/libpointmatcher/pointmatcher/DataPointsFilters/CovarianceSampling.cpp:265:17:   required from here
/home/duongthao/catkin_ws/src/StaticMapping/libpointmatcher/pointmatcher/DataPointsFilters/CovarianceSampling.cpp:160:18: warning: variable ‘eigenVa’ set but not used [-Wunused-but-set-variable]
  160 |  const Vector6   eigenVa = solver.eigenvalues().real();
      |                  ^~~~~~~
/home/duongthao/catkin_ws/src/StaticMapping/libpointmatcher/pointmatcher/DataPointsFilters/CovarianceSampling.cpp: In instantiation of ‘void CovarianceSamplingDataPointsFilter<T>::inPlaceFilter(CovarianceSamplingDataPointsFilter<T>::DataPoints&) [with T = double; CovarianceSamplingDataPointsFilter<T>::DataPoints = PointMatcher<double>::DataPoints]’:
/home/duongthao/catkin_ws/src/StaticMapping/libpointmatcher/pointmatcher/DataPointsFilters/CovarianceSampling.cpp:266:17:   required from here
/home/duongthao/catkin_ws/src/StaticMapping/libpointmatcher/pointmatcher/DataPointsFilters/CovarianceSampling.cpp:160:18: warning: variable ‘eigenVa’ set but not used [-Wunused-but-set-variable]
make[1]: *** [CMakeFiles/Makefile2:223: CMakeFiles/pointmatcher.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```
How to fix it?

EdwardLiuyc, 2022-03-03:
@duongthao1218 I saw just warnings, no errors in building. Maybe you have changed the default settings of your compiler?

duongthao1218, 2022-03-04:
I built repo successfully, thank for u response.

closed 2022-03-04 by EdwardLiuyc