EdwardLiuyc/StaticMapping #66: nothing is saved after process

lvjunning1243015876, 2022-03-03:
Dear Liu. Thanks for your nice work. I meet some problem. And there is only a empty `path.csv` in folder `pcd` after processing.
Data is collected by uav. And I have set `accumulate_cloud_num` to `360`.
Running result is below.
`
[ map_builder.cc: 716 ] [INFO] Finishing Remaining Computations...
[ map_builder.cc: 396 ] [INFO] point cloud thread exit.
[ map_builder.cc: 665 ] [INFO] **no enough frames for new submap, quit**
[ map_builder.cc: 712 ] [INFO] submap processing done.
[ map_builder.cc: 566 ] [INFO] All submaps have been connected.
[ map_builder.cc: 616 ] [INFO] generating path files ...
[ memory_manager.cc: 62 ] [INFO] End managing memory.
`

more
`./mapping_lidar_only.sh 
[ static_mapping_node.cpp: 112 ] [INFO] Get point cloud from ROS topic: velodyne_points
[ static_mapping_node.cpp: 126 ] [WARNING] No imu topic, expect no imu messages.
[ static_mapping_node.cpp: 146 ] [WARNING] No odom data accessed!
[ static_mapping_node.cpp: 161 ] [WARNING] No GPS data accessed!
[ map_builder.cc: 152 ] [INFO] Got tf : tracking -> lidar 
  translation : (0.04, -0.02, 0.3)
  rotation    : (0, -0, 0)
[ map_builder_options.cc: 103 ] [INFO] Load configurations from file: ./config/lidar_only.xml

*********************** configs from file ***********************

                      whole_options.export_file_path -> pcd/
                      whole_options.map_package_path -> pkgs/test/
                       whole_options.odom_calib_mode -> [default] 1
                       whole_options.separate_output -> [default] false
            whole_options.output_direct_combined_map -> true
                           whole_options.output_mrvm -> false
                     whole_options.output_kitti_pose -> [default] false
                         whole_options.separate_step -> [default] 200

                 output_mrvm_settings.output_average -> false
                     output_mrvm_settings.output_rgb -> false
              output_mrvm_settings.use_max_intensity -> [default] true
                 output_mrvm_settings.prob_threshold -> 0.57
                 output_mrvm_settings.low_resolution -> 1
                output_mrvm_settings.high_resolution -> 0.1
                       output_mrvm_settings.hit_prob -> 0.55
                      output_mrvm_settings.miss_prob -> 0.48
                       output_mrvm_settings.z_offset -> 0
          output_mrvm_settings.max_point_num_in_cell -> 10

                              Creating filter: Range
                                          min_range_ -> 4.5
                                          max_range_ -> 75
                      Creating filter: RandomSampler
                                      sampling_rate_ -> 0.5

                                       options->type -> 6
                         options->accepted_min_score -> [default] 0.7
              front_end_options.accumulate_cloud_num -> 360
             motion_filter_options.translation_range -> 1
                   motion_filter_options.angle_range -> 1.5
                    motion_filter_options.time_range -> 0.25
                  motion_compensation_options.enable -> true
             motion_compensation_options.use_average -> true
                                 imu_options.enabled -> false
                                    imu_options.type -> 0
                               imu_options.frequency -> 200
                        imu_options.gravity_constant -> 9.8
                                       options->type -> 6
                         options->accepted_min_score -> 0.7
                          submap_options.frame_count -> 2
                    submap_options.enable_inner_mrvm -> [default] false
                  submap_options.enable_voxel_filter -> false
              submap_options.enable_random_sampleing -> false
                 submap_options.random_sampling_rate -> 0.5
                           submap_options.voxel_size -> [default] 0.1
                   submap_options.enable_disk_saving -> true
                         submap_options.enable_check -> false
                    submap_options.disk_saving_delay -> 30
                   submap_options.saving_name_prefix -> s_
                     isam_optimizer_options.use_odom -> false
                      isam_optimizer_options.use_gps -> false
             isam_optimizer_options.output_graph_pic -> [default] false
       isam_optimizer_options.enable_extrinsic_calib -> [default] true
          isam_optimizer_options.gps_factor_init_num -> [default] 25
       isam_optimizer_options.gps_factor_sample_step -> [default] 1
    isam_optimizer_options.gps_factor_init_angle_rad -> [default] 1.6
                       loop_detector_setting.use_gps -> false
                loop_detector_setting.use_descriptor -> true
          loop_detector_setting.output_matched_cloud -> false
      loop_detector_setting.trying_detect_loop_count -> 1
         loop_detector_setting.loop_ignore_threshold -> 20
       loop_detector_setting.nearest_history_pos_num -> 5
       loop_detector_setting.max_close_loop_distance -> 5
     loop_detector_setting.max_close_loop_z_distance -> 2
              loop_detector_setting.m2dp_match_score -> 0.975
       loop_detector_setting.accept_scan_match_score -> 0.8
                          map_package_options.enable -> false
                   map_package_options.border_offset -> 100
                     map_package_options.piece_width -> 500
               map_package_options.cloud_file_prefix -> part_
               map_package_options.descript_filename -> map_package.xml

*****************************************************************

[ map_builder.cc: 63 ] [INFO] Init scan matchers.
       dist_outlier_ratio -> 0.7
      knn_normal_estimate -> 7
            max_iteration -> 100
[ map_builder.cc: 76 ] [INFO] Init isam optimizer.
[ map_builder.cc: 218 ] [INFO] Add a new trajectory : 0
[ map_builder.cc: 102 ] [INFO] Init threads.
[ map_builder.cc: 111 ] [INFO] Init finished.
[ map_builder.cc: 783 ] [INFO] disable odom.
[ map_builder.cc: 792 ] [INFO] disable gps.
[ memory_manager.cc: 53 ] [INFO] Start managing memory.
processed 8.042402 of 746.962851 seconds of the bag... 
...
processed 741.131710 of 746.962851 seconds of the bag... 
[ map_builder.cc: 716 ] [INFO] Finishing Remaining Computations...
[ map_builder.cc: 396 ] [INFO] point cloud thread exit.
[ map_builder.cc: 665 ] [INFO] no enough frames for new submap, quit
[ map_builder.cc: 712 ] [INFO] submap processing done.
[ map_builder.cc: 566 ] [INFO] All submaps have been connected.
[ map_builder.cc: 616 ] [INFO] generating path files ...
[ memory_manager.cc: 62 ] [INFO] End managing memory.

               block name |    times |  avg.(ms) |   sum.(ms) |  min.(ms) |  max.(ms)
-------------------------------------------------------------------------------------
                     main |        1 |    177011 |     177011 |    177011 |    177011
`

EdwardLiuyc, 2022-03-04:
@lvjunning1243015876 
I suppose you may have some misunderstanding on `accumulate_cloud_num`, this parameter means how many RAW point clouds you want to be merged as 1 point cloud. As you set it to 360 (if I get it right, you are using velodyne and the frequency of point cloud message is 10Hz), then every 36s you will get 1 point cloud to do mapping, which I will not do.
I suggest that you try with `accumulate_cloud_num` set to 1.

lvjunning1243015876, 2022-03-05:
Thanks very much. But when I set `accumulate_cloud_num` to 1, it still informs  `no enough frames for new submap, quit`.

EdwardLiuyc, 2022-03-06:
My guess is that you set the wrong frame_id or topic_name for lidar, or the input filter removed most of the points.  Could you please share me the rosbag you use so I can also check? (Through google drive or some other cloud drive).

lvjunning1243015876, 2022-03-08:
Yes, this is google driver link https://drive.google.com/file/d/1YSiRurMOfZEVrI21AmGO0ZRYU4bCKBvK/view?usp=sharing.

EdwardLiuyc, 2022-03-08:
@lvjunning1243015876 
I tried with your data but unfortunately I didn't got a good result from it. Regarding your problem, actually the `POINT_CLOUD_TOPIC` should be set to `/velodyne_points`. 

As your use case, I am not very confident if you should use this project. 
First is that this project is mainly tested with dataset collected by vehicles, so roll and pitch angle are always changing slowly. As far as I know, UAV senario is not friendly to Lidar SLAM becuase the high angle velocity will cause large distortion on raw point cloud which is really hard for motion compensation. For that, I also tested with imu data in your rosbag. The result is still a little bit disappointing. 
Secondly, your raw data shows really little difference in time. See images below. This is like a hallway problem. Wrong match happens a lot.
[image]
[image]

My suggestion (if you are still interested in this project):
1. Test your setup in a more feature-rich senario.
2. Or design a feature-rich path for data collection.
3. Achieve hardward level synchronization with lidar and imu.

Still, I will keep investigating with your data. :blush:

lvjunning1243015876, 2022-03-09:
Thank you for your patience and suggestions very much. And I will try other way to map.

closed 2022-03-09 by lvjunning1243015876