EdwardLiuyc/StaticMapping #67: Check failed: tf_buffer.canTransform(target_frame, source_frame, ros::Time(0), &tf_error_msg)

duongthao1218, 2022-03-08:
When I run `./mapping_lidar_imu.sh`, program failed. How can I fix it?
[image]

EdwardLiuyc, 2022-03-08:
If you notice the 5th line of the `mapping_lidar_imu.sh`, there is a file `urdf/test2.urdf` loaded by the main process to set transforms among different frames. You may need to adjust this file with your frame ids. Here is a [reference for urdf](http://wiki.ros.org/urdf/Tutorials/Building%20a%20Visual%20Robot%20Model%20with%20URDF%20from%20Scratch).

Next to that, I will add the support for listening TF broadcast soon.

duongthao1218, 2022-03-09:
@EdwardLiuyc Thank u so much. 
After process finished, I got a problem about my map. My map didnt show enough pointcloud that I collected.
I use Velodyne - VLP16, ardupilot cube (IMU data) and collect dataset indoor.
Information of my bag file, [drive link](https://drive.google.com/file/d/1lLTqVkhfOLlbN_1vWsM3xdBIBtbKHf_7/view?usp=sharing)
[image]

I used CloudCompare and this is result.
[image]

EdwardLiuyc, 2022-03-12:
@duongthao1218 
Sorry for the late response. This is what I got from your data (I tried first without imu), If you still want to use imu, you may need to adjust the parameters for imu, e.g: frequency:
[image]

EdwardLiuyc, 2022-03-12:
@duongthao1218 
This is what I got when mapping with lidar & imu, as you can see no big difference with the other one above. You lidar never moved so I assume imu isn't vital here.
[image]

BTW, seems that I really need to add a wiki for the parameters.

duongthao1218, 2022-03-14:
@EdwardLiuyc  Thank you for your response. After I ran  `./mapping_lidar_only.sh` and `rosbag play <path to my bag file>` , the output is not similar to the result that you showed me above. I'm sure that I changed lidar and imu topic name correctly.

EdwardLiuyc, 2022-03-14:
@duongthao1218 remember to set `accumulate_cloud_num` to 1  in the config xml file.

duongthao1218, 2022-03-14:
@EdwardLiuyc yea, I set `accumulate_cloud_num` to 1 in the config xml file before but there was no change
[image]

EdwardLiuyc, 2022-03-14:
@duongthao1218 So did you get a whole_map.pcd, could you check that? The output for static_map.pcd is disabled by default.

duongthao1218, 2022-03-14:
@EdwardLiuyc  Oh, I checked path.pcd but not whole_map.pcd.
Thank u so much. I apologize for my shortcomings.

closed 2022-03-14 by duongthao1218