EdwardLiuyc/StaticMapping #68: Adding a second lidar to system?

MGforest, 2022-08-03:
Hi, 
thank you for your great tool and documentation!
My question is, if it is easy to adjust the code, so that I can add a second lidar pointcloud source. On my system I have two lidar scanner and one imu. 

I hope to hear from you soon, have a nice day!

EdwardLiuyc, 2022-08-20:
Hi @MGforest , sorry for the late reply. I am just back from my holiday!
Actually this has been one of the TODO items for a long time. I don't think it's would be TOO hard to achieve this, which may cost a few days, maximum a week. I will try to add this feature soon but I don't have test datasets. Could you share me some data via google drive or some other cloud drive?

MGforest, 2022-08-22:
No problem, I hope you had a great holiday. 
Thank you very much in advance for your time, I am happy to share with you my data. 

In this [link](https://hessenbox.tu-darmstadt.de/getlink/fiJ4dcowofmDYbxdmi7nCMBQ/example_2_lidar.bag) I share with you a bagfile that I collected with 2 Velodyne VLP16, one IMU and one GNSS antenna. 

the urdf file is as follows: 
```
<robot name="backpack">
  
  <link name="base_link">
    <visual>
      <geometry>
        <box size="0 0 0"/>
      </geometry>
    </visual>
  </link>
  
  
  <link name="vlp201">
    <visual>
      <geometry>
        <cylinder length="0.2" radius="0.2"/>
      </geometry>
    </visual>
  </link>

  <joint name="base_to_velo" type="fixed">
    <parent link="base_link"/>
    <child link="vlp201"/>
    <origin xyz="-0.02 0 0.830" rpy="0 0.785 1.571"/>
  </joint>
  
  <link name="imu_link">
    <visual>
      <geometry>
        <box size="0.1 0.1 0.05"/>
      </geometry>
    </visual>
  </link>

  <joint name="base_to_imu" type="fixed">
    <parent link="base_link"/>
    <child link="imu_link"/>
    <origin xyz="0.0 0.0 0.0" rpy="0 0 3.142"/>
  </joint>
  
  
</robot>
```
For the second VLP16 the parameters are: 
 ```
 <link name="vlp202">
    <visual>
      <geometry>
        <cylinder length="0.2" radius="0.2"/>
      </geometry>
    </visual>
  </link>

  <joint name="base_to_velo" type="fixed">
    <parent link="base_link"/>
    <child link="vlp202"/>
    <origin xyz="-0.37 0 -0.15" rpy="0 -1.57 -3.142"/>
  </joint>
```

MGforest, 2022-09-02:
Hi @EdwardLiuyc, 
have you already tried if it works?

closed 2023-03-02 by EdwardLiuyc